czipis/mavic-mini-battery-info #1: Should it work with the mavic mini 2 batterys? they have the same connector.

Travisco29, 2020-11-18:


bzushi, 2021-01-25:
I just finished up this project tonight, and I confirm it works with the DJI Mini 2 batteries.

spanzetta, 2021-04-23:
> 
> 
> I just finished up this project tonight, and I confirm it works with the DJI Mini 2 batteries.

Hi,
where did you find the battery connector?
It's the only component missing from the list..
I need the one for Mavic Mini 1 Battery
Any help would be appreciated
Thanks

bzushi, 2021-04-23:
I had a bunch of JST headers, and used a 6pin female for this. Something like this will probably work for you: https://www.amazon.com/2-0MM-Female-Single-Connector-Wires/dp/B01IZWYK7I

jan-cipa, 2021-04-23:
be sure pin header has 2.0mm spacing (not the standard 2.54mm)

dad401, 2022-01-06:
I also confirm that it works with Mini 2 batteries. 
Despite the use of JST 2.0mm connectors, the connection is sometimes a bit tricky/unstable. If you only see a white screen after the init screen it is often caused by a bad connection from your battery to the Arduino. Check your cables and move your battery slightly on the connector.
To get Mini 2 batteries to work you have to "activate" them in the charger (case). Just put them into the charger and press the side button. After this (don`t wait too long) you can use them with the Nano/display without a connection to your PC.